frazew/VirtualSensor #2: Android rotation vector stops working.

ff2009, 2016-07-27:
When  I enable the module, on xposed, after rebooting the phone my android rotation vector doesn't work anymore. 
Disabling the module a rebooting the phone makes it work again. My phone is an lg-l90.

Frazew, 2016-07-27:
That's weird. Could you upload the xposed log just in case if possible ?

ff2009, 2016-07-27:
Here it is.
By the way. I noticed that the mix of sensors is not very accurated.
When I through the phone up in the air rotating when  he fall he doesn't know in witch position he was before. Is it possible to assign the android rotation vector to the gyroscope.
I use the sensor fusion to do this test.

Frazew, 2016-07-29:
Latest version (v1.31) should fix this. If I understood the problem correctly, what happened with previous versions is that existing sensors were sometimes replaced by the virtual sensors. This should now be fixed. Could you test and report any further issues ?

ff2009, 2016-07-29:
I have tried the new update, with the version (v1.32) on 2 different devices, and I will send you the xposed logs, and some screen shots, from the CPU-Z on the sensors tab, with the module enable and disable.

LG-L90:
 module enable:
- the google camera show the panorama and photosphere options and both work (but the sensors are very inaccurate) (It's impossible to take a complete photo sphere).
- On the sensor fusion app, the android rotation vector still, not working. (I know it still a early build, but the virtual sensor, on sensor fusion, lost track, of it's orientation, when, I through the phone around.
- On CPU-Z, only the accelerometer, and the magnetometer work.
- On Just Cause 3 wingsuit app, the gyroscope worked on the last version 1.2, I think (very inaccurate, but I could see, that was doing something), and now doesn't do nothing.

Wiko Wax:
 module enable:
- The CPU-Z show all virtual sensors enable, and they seem to work.
- On Sensor Fusion, the android rotation vector isn't on the default position, but it's static. The gyroscope is on the default position, and don't move either.

By the way, took a look at this project if you hadn't already:
https://bitbucket.org/apacha/sensor-fusion-demo/
 
I hope I could help.

[image]
[image]
[image]

[image]
[image]

[lgl90_xposed_error_20160729_152144.log.txt](https://github.com/Frazew/VirtualSensor/files/390803/lgl90_xposed_error_20160729_152144.log.txt)
[wax_xposed_error_20160729_151248.log.txt](https://github.com/Frazew/VirtualSensor/files/390802/wax_xposed_error_20160729_151248.log.txt)

Frazew, 2016-07-29:
First of all, thanks for the very detailed information !

The log for your lgl90 has an exceptions I did not expect to find, it's very useful. This explains why your device does not detect the sensors, they don't seem to be correctly added to the sensors list. I think the issue is that your device already has quite a lot of sensors this mod is supposed to add, I thought I had a check not to override existing sensors, but it seems it doesn't work very well. 

As for the wax log, two possibilities : the main exception thrown on many other devices is the issue. Or there is indeed a particular issue that causes the magnetic to stop reading anything. I can see that on your sceenshot indeed : the magnetic sensor doesn't output anything and so do the custom sensors because of that.

ff2009, 2016-07-30:
I tested this release on the LG L90, and seems that fix the error.

The logs doesn't seem to have anything relevant, but here it's:

ff2009, 2016-07-30:
Just finish testing on the Wiko Wax, and the log doesn't seem to had errors related with the your module.

Log:
[wax_xposed_error_20160730_134340.log.txt](https://github.com/Frazew/VirtualSensor/files/391915/wax_xposed_error_20160730_134340.log.txt)

Wiko Wax:
- On CPU-Z some sensors apear with some values, but other only show a small diference between values, ranged from 0.0, -0.5. The only one that show some relevant data is the Gyroscope.
- On sensor fusion, all sensors are static, but only the gyroscope dependents, seems  to be doing something, because they aren't in their default position, when I launch the app. but they don't move. All the other sensors are static and in their default position. Even the compass and Accelerometer mix, but on CPU-Z, the values are fine.
I couldn't test the Google Camera because it doesn't work, on android 4.4 anymore.

Frazew, 2016-07-30:
Thanks you for the reports. It's good to see that at least the sensors no longer get overridden.
Many people have issues with actually getting the sensors to report any values at all. I still haven't figured out why it doesn't work. From what you're saying, it might be a range issue.

closed 2016-08-06 by ff2009