frazew/VirtualSensor #3: A little bug of significance

Prostokuche, 2016-07-27:
The gyro doesn't rotate but I'll admit that it works smoother than the previous version. A bit slow but very smooth. I'm using marshmallow/ Xposed v86 /Huawei p8 lite. Thank you for the hard work.

Prostokuche, 2016-07-27:
Oh. And I can't turn left and right.

Frazew, 2016-07-29:
Thanks for the report. Could you add a little more details ? What app did you test the gyro with ? Is the issue still there with the latest (v1.31) version ?

ff2009, 2016-08-06:
I have the same problem here I can look up and down, and tilt, but I can't look to the right or left.
But this only happens on just cause 3 Wingsuit. 
On the google camera in panorama ou photosphere mode, I can turn left and right. Probably it uses a diferente sensor.
And yes I am using the last version.

Frazew, 2016-08-07:
See my answer here : #13 for instance. All I can say is : you're not alone. I'm working on it.

Frazew, 2016-08-07:
Like I just commented on the other issue, please try this version. It might (hopefully) fix this issue.
https://github.com/Frazew/VirtualSensor/releases/tag/v1.5-test

okejokej, 2017-02-27:
Using a TP-Link Neffos C5 Android 5.1 device (has a compass and an accelerometer).
Unfortunately the module doesn't really work with much as it is.  Version 1.5 doesn't improve anything in my case.

The gyroscope is recognized, allowing to install software that requires it, also it returns some values in its own gui. I noticed in there that it mostly focuses on accelerometer though, as the values ever only reach values above 1 when there's movement, and otherwise they're oscillating around 0 - is that how a phone gyro works?
I didn't say it doesn't work at all, because it actually works with an older version of Google Cardboard - 1.5 (and possibly older). It's very, very jittery and doesn't do very well on vertical movement, but it clearly comes from your module as there's nothing at all when I disable it - so yeah, it certainly works for some methods of apps getting gyroscope data.
I hope you are still interested in this project and find time to work on it eventually - as I'm certain it is possible to get pretty good, non-jittery gyro emulation. Because Google Maps app (as well as Street View app, but it's kinda buggy there) has it.

You probably know this, but when you go to street view mode in Google Maps there's a little compass in top right, it actually turns on phone position tracking. I don't know if it uses compass alone or compass+accel combo like your module, but it is actually very smooth, accurate and works well in both horizontal and vertical movements. There's a tiny delay, possibly due to smoothing, but it probably could be dealt with by allowing some light jitter. 
Unfortunately this head tracking option is not there in VR mode in the Street View app, possibly the only reason for that is that google's own Cardboard headset would be useless there with its magnet messing with compass readings :/

Cheers! :) Hope you can find time for this between your daily life and college.

okejokej, 2017-02-28:
There also seems to be a little bit of confusion amongst reports of what works with the Virtual Sensor. I've seen people claiming that it works with VR app "VR Cave", whereas it does not use the module, it has its own compass head tracking (that works really well), that doesn't require Xposed or even root. Another app that has its own gyro emulation is "Sites in VR", but it's very wobbly and jittery and not very usable in there. It also seems that the app doesn't work anymore with Xposed installed for some reason.