frazew/VirtualSensor #4: It does not work on the Galaxy Grand 2

GuhPlays, 2016-07-28:
Good wanted to thank you for creating this module but it does not work on my phone has to support it? Android is 4.3 jelly bean.

Frazew, 2016-07-29:
Could you try with the newest version (v1.31) ? If that doesn't work, could you provide the xposed log too ?

GuhPlays, 2016-07-31:
Yes I will do it now after the results speak.

GuhPlays, 2016-07-31:
I tried it with the latest version and still did not work, here's the log http://pastebin.com/bVbBDqPu

Frazew, 2016-07-31:
Thanks for the log, I don't know why it doesn't work though as the hook should be ok. I'm really starting to think that Samsung devices have something different.

GuhPlays, 2016-08-01:
Oh,ok I'll wait.

Frazew, 2016-08-01:
Does it work with the latest version though ?

GuhPlays, 2016-08-01:
[image]
Yeah !! It worked but still could not see 360 degree videos on youtube.

closed 2016-10-03 by Frazew