frazew/VirtualSensor #5: Add gyroscope when no magnetic sensor

duramato, 2016-07-29:
My phone has no or a non working magnetic sensor/compass,  but I would still would like to have a fake/virtual gyroscope. 
Would it be possible to emulate the info only from the accelerometer?

Frazew, 2016-07-29:
Unfortunately this is impossible. Not that it is too complicated or that I don't want to do it, it *is* actually impossible.

The accelerometer only allows for detection of wether you're moving the phone up or down and/or tilting but the rotation as we generally understand it (ie moving the head left/right) is only possible thanks to the compass too.

closed 2016-07-29 by Frazew

Mark42XLII, 2017-07-12:
But could you make it just so it looks up/down? I planned in putting an external compass to work...