frazew/VirtualSensor #6: Pokemon GO wont recognize Sensor

derubm, 2016-07-31:
The Vsensor works fine on Huawei P8 Lite, all sensor test tools i tried can find it, and it works. Pokemon go However wont recognize the Sensor :( - Anyway fine work, you might find a way to make it also work for Pokemon GO :)

Frazew, 2016-07-31:
I'm aware of this issue but I haven't yet found any explanation as to why it doesn't work. I'm still working on it though !

TurokO8A, 2017-04-08:
Hey ^^ great module! Any news/ETA on this?

Frazew, 2017-04-17:
If only I had enough time to work on it ! I initially wanted to publish a release only after getting the cardboard hooks to fully work but I guess I'll release something that includes the many fixes missing in the previous releases and the wip hooks for cardboard as a toggleable option.