frazew/VirtualSensor #11: It does not work on the SM-G530BT

zezinbm, 2016-08-05:
tested it on my SM- G530BT and did not work , it supports android 4.4.4 ?
sorry for the mistakes , I'm posting with the google translator

Frazew, 2016-08-06:
Could you give more details ? Is the gyroscope detected but not giving any values or is it not working at all ? Can you send the xposed log ?

zezinbm, 2016-08-06:
this is the problem I'm half lay in this part , I am searching here for how to do to put the log here

zezinbm, 2016-08-06:
[image]

zezinbm, 2016-08-06:
fiz a instalacao do xpoded , reiniciei o aparelho e funcionou, obrigado pela ajuda, pode trancar o topico

zezinbm, 2016-08-06:
he worked in gyrotest , managed to install the card board , but in the games he did not budge

zezinbm, 2016-08-06:
I did a hard reset on the device, I installed Xposed after the virtual senssor , and nothing worked , the app opens , appears the indications of senssor , but the game works not

Frazew, 2016-08-06:
From the screenshot you sent, it appears you did not install the correct Xposed version for you android version. If this is really the case, you will need to fix this in order for the module to work.

zezinbm, 2016-08-06:
I managed to install the current version of Xposed before I could install the card board after putting the virtual sensor, now not even that, all gyroscope testing application work , not only works in games, bombsqad type , if not meche

zezinbm, 2016-08-06:
[image]
[image]
[image]
[image]

Frazew, 2016-08-07:
I guess that means I can close this issue then.

closed 2016-08-07 by Frazew