frazew/VirtualSensor #12: "Unfortunately, VirtualSensor has stopped"

conceptodaf, 2016-08-06:
Hi @Frazew , thank you very much for your efforts! I just installed your apk and this message appeared after I tried to launch it. I'm not sure if it's only developed for phones or any device with Android. I tested it on my tablet: Samsung Galaxy Tab 2 7.0 (GT-P3110), Cyanogenmod 13 (Android 6.0.1). If I can provide any more details just let me know. And thanks again!

conceptodaf, 2016-08-07:
**Update:** I'm so sorry, I'm new at this and I didn't know I had to install Xposed first (I didn't even know what Xposed was until I found a thread on XDA haha). Maybe this is useful for someone as newbie as I am: 

I installed the Xposed apk file but it said it wasn't active so I uninstalled it and tried installing the file "xposed-v86-sdk23-arm.zip" using TWRP instead. I cleaned the dalvik cache and rebooted the system.
I thought it was going to be shown in the app list but I was wrong, so I thought something went wrong and installed the apk again. This time it showed me a message in green which said "Xposed framework version 86 is active". 
After that, I installed the VirtualSensor and I enabled it from Xposed. I tried to open it but it showed me again the "Unfortunately, VirtualSensor has stopped" message. I rebooted again the system and now it seemed to be working, I could open it and it showed me a lot of numbers. But I downloaded Sensor Box for Android to check and it says I have a giroscope now, but when I move my device nothing happens. Unfortunately, in my case it doesn't work on Pokemon Go either.

zezinbm, 2016-08-07:
I am not abusing the developer of course , I thank him for having made ​​this application , but if it developed like a launcher for this app can run in the games, will be a big step

Frazew, 2016-08-07:
So you're not getting any values at all ? At first I would have said that it was Samsung ROM's fault but as you're running cyanogenmod, that's not the issue.
As for Pokémon GO, this is a known issue

conceptodaf, 2016-08-07:
Thanks for your reply @Frazew! VirtualSensor shows me some values (they change really fast) but nothing else happens, I tried watching a 360º video on youtube and it didn't work out:
[image]

And this is what Sensor Box for Android shows, the box doesn't move at all:
[image]

And again, thanks a lot for your hard work!

Frazew, 2016-08-07:
It shows some values, that's already a good thing ! According to your screenshot, the theoretical and "real" values are the same, which means that the module works fine. YouTube 360 doesn't work fine just yet as it uses the Carboard API which this module doesn't hook as of now. As to Sensor Box, it always acted in a weird way for me. It either didn't move or move very slowly, I think it might apply some kind of filter to the values. You could test the module with "Sensor Fusion Demo", which works much better and allows to compare the gyroscope's output to other sensor's output (The app does sensor fusion). Let me know if you find anything interesting !

conceptodaf, 2016-08-10:
Hi, @Frazew! Thanks a lot for your reply. You were completely right, I downloaded and installed Sensor Fusion and it works! Some sensors work better than the others, but it's a great progress. I've been trying for the past two days to record the screen of my tablet while I test the sensors because I thought it would be cool for you to see another device working, but unfortunately I couldn't. I tried with many apps but they crash all the time (if you know one which makes that kind of task, let me know and I will gladly give it a shot). I saw you uploaded a new version of VirtualSensor but I didn't install it yet, I will let you know when I do it and how it's working. Thanks again for your efforts!

Abhishek9634, 2016-09-17:
Same issue in Moto X Play

JaredTheWolf, 2016-09-28:
@Frazew Thanks foe the updates on this. Ive been trying to get FullDive, Pokémon GO and YouTube to all work. Interestingly, regular YouTube workes at first but then stopped detecting motion (in Cardboard mode, i suspect ths was an update to YouTube that did this). Will you be working on hooking Cardboard, or is that beyond what this project is right now?

Frazew, 2016-09-28:
I'm working on it actually. Although I don't have much time (studies take a lot of it), I regularly try to see what I can do to hook Cardboard. I know where to hook (in fact, part of the hook is already in the code here on Github), but I can't get it to work. Hopefully though I'll figure this out somehow and make everything work.

JaredTheWolf, 2016-09-28:
Great to hear! Thanks for all your amazing work. Will be donating soon ^^

Frazew, 2016-10-03:
Issue should now be fixed as of commit #09a04d02e8d97379e084f98415a9878e8cef688d .

closed 2016-10-03 by Frazew