frazew/VirtualSensor #13: Can't move right or left with Zenfone 2 (Z008) CM13 Nightly

ninhvuhungkhang, 2016-08-07:
Hello Frazew, first of all i'd like to thank you for VirtualSensors is a nice project for all those lack of Gyroscope sensor or malfunction Gyroscope. I have test it on sample videos on YouTube that it work well while I hold my phone up and down. But it doesn't work while rotate my phone left to right that it standing still and finally, Pokémon GO seems won't detect the VS as well. Can you see a problem to it? Thanks

P/S I'm using Zenfone 2 Z008 (720p) with CM13.

Frazew, 2016-08-07:
Thanks for the report ! Many people have this issue, I'll link the same issues together in order to better narrow the problem down.

ninhvuhungkhang, 2016-08-07:
Ok, hope you'll fix this issue soon! Anyway, do you need further information about my phone or something else?

Frazew, 2016-08-07:
I'm not sure, it happens on a variety of devices so I don't think I need any more information. I actually thought of a possible fix (that I cannot test for now). I'll post something back here when a test build is ready.

Frazew, 2016-08-07:
Could you try with this test release and report back any progress (or not if it doesn't change anything). Thanks ! 
https://github.com/Frazew/VirtualSensor/releases/tag/v1.5-test

ninhvuhungkhang, 2016-08-08:
The problem still not fixed, still can't rotate left/right and i don't know what happened but after install new version, it still show that i'm in version 1.41

ff2009, 2016-08-08:
Yup. Still does't rotate right or left, and the movement is not so smooth now. I have a little freezes when moving the phone around, but this could be my device.

LOG:
[lgl90_xposed_error_20160808_152103.log.txt](https://github.com/Frazew/VirtualSensor/files/406995/lgl90_xposed_error_20160808_152103.log.txt)

ninhvuhungkhang, 2016-08-08:
ff2009 I think you should update to Xposed version 86 to see if there is something different?

ff2009, 2016-08-08:
My bad. I knew that something wasn't right with my xposed, but I didn't noticed that. 
I tried the xposed update, and I can't turn right or left still. But I get ride of the freezes. So thanks.

ninhvuhungkhang, 2016-08-09:
But still hope that the turn left/right problem got fixed

ninhvuhungkhang, 2016-08-15:
Het Frazew, can you observe the 360 video of Facebook? I can tilt left right, up down as normal without gyroscope. Maybe you can use that as reference