frazew/VirtualSensor #15: Xperia m4 aqua

diyari95, 2016-08-13:
Hi my friend thanks for doing this hard work, my phone doesn't have gyroscope so i tested your work now my phone support gyroscope but still three axis are not working i tested from cpu x app, if you need logcat file i can give it to you just fix it thanks

Frazew, 2016-08-14:
I think CPU-Z doesn't work well with this module. I'll have to look at how they implement the sensor readings in order to understand. Could you test with Sensors Test for example and tell me if that changes anything ?

diyari95, 2016-08-14:
Hi thanks for answering me i tested with sensors test app and result is not
work :-(

diyari95, 2016-08-14:
On my Gmail i sent pic but it doesn't appear here

diyari95, 2016-08-15:
I tested with 360 YouTube video, up and down works but left and right is not working. And please support for Pokémon go,

diyari95, 2016-08-16:
Update my state 
Hi again i tested with sensor kinetics  and gyroscope is working perfectly in this app even left and right  works  i hope it helps you

Frazew, 2016-08-16:
Thanks ! It is indeed supposed to work. I'm still working on Cardboard (aka Youtube 360) and Pokémon GO

diyari95, 2016-08-16:
Thanks again for your hard work, i hope to see Pokémon go support very soon.