frazew/VirtualSensor #16: Gyroskop sensor shopws only 0.00 as value - error 'invalid value for field' in Xposed-Log

Starfiresg1, 2016-08-25:
After installing this the "Virtual Sensor" is visible but only shows value 0.00 (in Sensor Kinetics and as theoretical values in Virtual Sensor App):

Screenshots:
[https://postimg.org/image/nb02ffsox/](url)
[https://postimg.org/image/8d4t3mpvl/](url)

Also Xposed-Logfile shows an error (invalid value for field) - here is the complete log:

`-----------------
25.08.2016 12:04:39 UTC
Loading Xposed v54 (for Zygote)...
Running ROM 'ALPS.JB2.MP.V1.2' with fingerprint 'WIKO/wiko/wiko:4.2.1/JOP40D/1434680398:user/release-keys'
Loading modules from /data/app/com.devadvance.rootcloak2-1.apk
  Loading class com.devadvance.rootcloak2.RootCloak
Loading modules from /data/app/biz.bokhorst.xprivacy-2.apk
  Loading class biz.bokhorst.xprivacy.XPrivacy
Loading modules from /data/app/com.ceco.gm2.gravitybox-2.apk
  Loading class com.ceco.gm2.gravitybox.GravityBox
GB:Hardware: mt6589
GB:Product: wiko
GB:Device manufacturer: Enspert
GB:Device brand: WIKO
GB:Device model: DARKNIGHT
GB:Device type: phone
GB:Is MTK device: true
GB:Is Xperia device: false
GB:Has Lenovo custom UI: false
GB:Has telephony support: true
GB:Has Gemini support: true
GB:Android SDK: 17
GB:Android Release: 4.2.1
GB:ROM: ALPS.JB2.MP.V1.2
GB:PatchFakeId: Patching FakeID vulnerability
GB:PatchMasterKey: Patching MasterKey vulnerabilities
Loading modules from /data/app/fr.frazew.virtualgyroscope-1.apk
  Loading class fr.frazew.virtualgyroscope.XposedMod
Loading modules from /data/app/de.robv.android.xposed.mods.appsettings-1.apk
  Loading class de.robv.android.xposed.mods.appsettings.XposedMod
VirtualSensor: Using version 1.41
java.lang.IllegalArgumentException: invalid value for field
	at java.lang.reflect.Field.setField(Native Method)
	at java.lang.reflect.Field.set(Field.java:588)
	at de.robv.android.xposed.XposedHelpers.setStaticObjectField(XposedHelpers.java:725)
	at fr.frazew.virtualgyroscope.hooks.SystemSensorManagerHook$API1617.afterHookedMethod(SystemSensorManagerHook.java:105)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:645)
	at android.hardware.SystemSensorManager.<init>(Native Method)
	at android.app.ContextImpl$30.createService(ContextImpl.java:487)
	at android.app.ContextImpl$ServiceFetcher.getService(ContextImpl.java:244)
	at android.app.ContextImpl.getSystemService(ContextImpl.java:1552)
	at de.robv.android.xposed.XposedBridge.invokeOriginalMethodNative(Native Method)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:631)
	at android.app.ContextImpl.getSystemService(Native Method)
	at android.view.WindowOrientationListener.<init>(WindowOrientationListener.java:78)
	at android.view.WindowOrientationListener.<init>(WindowOrientationListener.java:63)
	at com.android.internal.policy.impl.PhoneWindowManager$MyOrientationListener.<init>(PhoneWindowManager.java:600)
	at com.android.internal.policy.impl.PhoneWindowManager.init(PhoneWindowManager.java:902)
	at de.robv.android.xposed.XposedBridge.invokeOriginalMethodNative(Native Method)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:631)
	at com.android.internal.policy.impl.PhoneWindowManager.init(Native Method)
	at com.android.server.wm.WindowManagerService$3.run(WindowManagerService.java:859)
	at android.os.Handler$BlockingRunnable.run(Handler.java:745)
	at android.os.Handler.handleCallback(Handler.java:725)
	at android.os.Handler.dispatchMessage(Handler.java:92)
	at android.os.Looper.loop(Looper.java:153)
	at android.os.HandlerThread.run(HandlerThread.java:60)
java.lang.IllegalArgumentException: invalid value for field
	at java.lang.reflect.Field.setField(Native Method)
	at java.lang.reflect.Field.set(Field.java:588)
	at de.robv.android.xposed.XposedHelpers.setStaticObjectField(XposedHelpers.java:725)
	at fr.frazew.virtualgyroscope.hooks.SystemSensorManagerHook$API1617.afterHookedMethod(SystemSensorManagerHook.java:105)
[... 2850 lines cut ...]
	at fr.frazew.virtualgyroscope.XposedMod$13.afterHookedMethod(XposedMod.java:356)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:645)
	at android.hardware.SystemSensorManager.unregisterListenerImpl(Native Method)
	at android.hardware.SensorManager.unregisterListener(SensorManager.java:551)
	at abui.b(:com.google.android.gms:225)
	at abue.d(:com.google.android.gms:123)
	at abso.b(:com.google.android.gms:296)
	at abue.d(:com.google.android.gms:123)
	at abrc.run(:com.google.android.gms:15)
	at android.os.Handler.handleCallback(Handler.java:725)
	at android.os.Handler.dispatchMessage(Handler.java:92)
	at android.os.Looper.loop(Looper.java:153)
	at abur.run(:com.google.android.gms:250)
java.lang.NoSuchFieldError: android.hardware.SystemSensorManager#mSensorListeners
	at de.robv.android.xposed.XposedHelpers.findField(XposedHelpers.java:69)
	at de.robv.android.xposed.XposedHelpers.getObjectField(XposedHelpers.java:611)
	at fr.frazew.virtualgyroscope.XposedMod$13.afterHookedMethod(XposedMod.java:356)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:645)
	at android.hardware.SystemSensorManager.unregisterListenerImpl(Native Method)
	at android.hardware.SensorManager.unregisterListener(SensorManager.java:551)
	at abui.b(:com.google.android.gms:225)
	at abue.d(:com.google.android.gms:123)
	at abso.b(:com.google.android.gms:296)
	at abue.d(:com.google.android.gms:123)
	at abrc.run(:co

Frazew, 2016-08-26:
Ah, never seen this one. Thanks for the report !

Frazew, 2016-09-10:
Fixed by commit #3f2e7062

closed 2016-09-10 by Frazew