frazew/VirtualSensor #20: Gyroscope Does Not work/move - Samsung Galaxy Grand Prime SM-G530W

JamesMacLean, 2016-12-22:
Hello Developers. Thank you for making this app. Unfortunately it is not working on my phone.

My phone:
Device Name: Galaxy Grand Prime
Model: SM-G530W
Android Version: 5.1.1
Virtual sensor version 1.41

[image]

I have installed the Xposed framework on my phone. The Xposed installer verifies that that the framework is installed. I have used an Xposed module called Device Faker (to get the vr apps to download from the play store) and this module works.  The Xposed framework is not the official version though. I am using an unofficial version designed for Samsung devices. If you try to install the default version it creates bootloops because Samsung modified android to include a feature called Touchwiz. The version I installed is detailed here: https://devs-lab.com/download-install-xposed-for-samsung-devices.html

[image]

I have enabled the VirtualSensor module in Xposed installer and rebooted. When I open the VirtualSensor app the changing gyroscope readings appear on the screen and the app acknowledges that i have an accelerometer and magnetic sensor. 
[image]

I have tried three apps that us gyroscopes and none of them worked:
1. Sensor Box for Android
This is an app you can use to test sensors on your phone. It acknowledges that I have a gyroscope, but whenever I move the phone and try to get the colored box to rotate, the box doesn't move.
[image]

2. Google Cardboard
The app works, but in the setup demo I can't rotate my view and look at the wolf. The view stays still.

[image]

3. FullDive VR
The app works, but I can't rotate my view. The view stays still

I seem to be the second user of a SM-G530 variant to have this problem. Let me know if you need any more information or any files.

Cheers,
James

JamesMacLean, 2016-12-22:
I have downloaded Android Studio and will try to debug and get a logcat to you in the next couple days.

iago318, 2016-12-23:
I also have a sm-g530m and the same thing happens to me

Lattoria, 2016-12-23:
Me too, Galaxy A7 2016 and VirtualSensor is unavailable in Google Cardboard and similar apps.

Frazew, 2017-01-07:
I don't have any samsung device to test the module on so it's harder for me to make it work on these devices. I had already looked into this issue a while ago and didn't find anything that would stop it from working (at least that's what I remember).

JamesMacLean, 2017-01-08:
@Frazew I would like to help somehow, but I am not an android developer. But I could probably get some sort of log file for you if you give me instructions. I have been playing with the android studio android monitor tool and there do seem to be some error messages when I open up apps that use a gyro.

fahririfkinaila, 2017-01-08:
I got a same problem, but with original Xposed Installer, my device is OPPO Neo 5 R1201

Frazew, 2017-01-08:
I think I'll make a "debug" module that will log the state of the hooks to pin down the issue. I'll update here when I have it.

perritochuchu, 2017-01-11:
hi, I'm test the module in a samsung prime sm-g531m, with "Sensor Kinetics Pro"  the gyroscope test work fine ,  but in vr apks or carboards apks doesn´t work.

jjgh, 2017-02-23:
Got the same issue with a Moto G (first gen). Your app shows gyroscope data being updated, but all apps using gyroscope will not detect any motion.

mangstad, 2017-02-27:
I don't have a Samsung device, I have a Blu Life One 4g LTE (x010q), however this is the same behavior I see with my phone. It has both an accelerometer and magnetic sensor. After installing the virtual gyroscope, activating in xposed, and rebooting, the phone passes any tests for gyroscope presence, but using it in any apps completely fails to move at all. Tried with both 1.41 and 1.5 test release.

Frazew, 2017-03-05:
I see you've tried with the "1.5 test release" (I guess that means you built it yourself from the source here) but which Android version do you run ? The latest fix I did (ie commit 437457376d0897a71d679e1e54b1212a9a014ade) was only for SDK23, I did not modify the hooks for older versions. I'll do it, it's just that I was testing things on my device, which runs on Android M.

mangstad, 2017-03-06:
Hmm, no I didn't compile myself. I just downloaded the apk from the v1.5-test pre-release on the "Releases" page on the repo. I'm on android 4.4.4

devian777, 2017-07-03:
Is there any progress about this? Im using grand prime g530h and facing the almost same problem. The app seems to be updating. But doesnt work with most vr app. There some app that can turn up down but very slowly but can't turn left/right. Is the left right using magnet sensor instead of compass?

Frazew, 2017-07-16:
I now have some time, but unfortunately no laptop to work on. I'll try to get a build ready because it's really been a while and there are some issues fixed (hopefully).
Indeed, left/right is using the compass. It seems to be the main issue everyone's having, we'll see if the next build fixes it (I remember fixing it, but I'm not sure about that).

Aptony, 2017-11-29:
Moto G (2015) user here, it works well for nougat I can watch 360 youtube vídeos can move  to everywhere the problemas is when I press the VR carboard button it loads smooth, i can watch the vídeo, cant pause or skip forward i dont know if its how it works, but cant move top, down, left or right, just like the report in this issue. Regards