frazew/VirtualSensor #29: Can't build on Android Studio.

Mark42XLII, 2017-07-12:
What SDK did you use to make this? This is not just possible. I've tried everything. I read like, 30 stackoverflow and nothing worked. Do I need some library or anything? Even formatted my whole PC and downgraded to Windows 7.

Error:

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugResources'.
> com.android.ide.common.process.ProcessException: Failed to execute aapt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
```

Frazew, 2017-07-16:
Hey, if I remember, this happenned to me too last time I tried to work on this project.
I'll see if it does it again and if I find a solution, I'll get back to you.

renantopac, 2017-07-31:
Hi, I faced the same issue, i solved creating a new project in android studio and then recreating the files of project one by one, probably after that you will get a error message like this "Failed to create MD5 HashFile" to solve this you need to navigate to C:\Users\'user'\ .android\build-cache and delete the 'xyz' folder, and then rebuild the project. 
'xyz' will be similar to c0695669ead81e6635797b952f69ee10fae82501

Mark42XLII, 2017-08-04:
I will try to do this when I find some time and get back the results here if it works.

Frazew, 2017-08-08:
Just so you know as this won't help you, I cloned the repository and built the project on a new Windows 10 PC on which I had just installed Android Studio. I had no issue.

okejokej, 2017-08-09:
Ah, you missed the opportunity to have exactly 1 year since last release :dancer:

Frazew, 2017-08-09:
Wow, it's actually been a year. I had absolutely no time to work on it this year and I only got access to a laptop recently (which is why I've been able to get back to working on this module)

imtlmzamora, 2017-09-03:
Windows 10, getting the same error; what could it be?

aayes89, 2022-08-30:
Finally, someone managed to compile the project? I have managed to do it with the latest version of Android Studio but after activating it in Xposed and rebooting, it does not enable the virtual sensor, however, when using the version already compiled, it does load the virtual sensor. Any reason why this happens?