frazew/VirtualSensor #30: Not working

gyro-needer, 2017-10-09:
Nothing works. I enter the app and I only see this, with no data.
[image]

Frazew, 2017-10-09:
Can you provide more details ? The Xposed log would be nice, but I think a logcat dump would be even more helpful.

Thank you !

gyro-needer, 2017-10-10:
[error.log](https://github.com/Frazew/VirtualSensor/files/1373040/error.log)
or
https://pastebin.com/5JZqWDrr

I'm on Android 4.4.4, my phone is a Samsung Grand Neo Plus (i-9060)

About the logcat, i don't know what is it/where to find it.

Frazew, 2017-10-11:
I found the issue, I was using a function that did not exist yet in Xposed v54. However, I'm not sure that this is the actual source of the missing data, the errors logged are linked to the cardboard hook and they should not interfere with the other hooks. We'll see.
Thanks for your help !

gyro-needer, 2017-10-11:
Thanks for reply, I hope you find the fix to it!

okejokej, 2017-11-14:
@gyro-needer The best fix you can use is to update your Xposed...

@Frazew when I was playing with this module I saw that indeed errors anywhere would generally affect the whole module, even if the affected code seemed unrelated to anything else in it.

That said, frankly I think that further developing this module is not worth your time. Gyro is being put into cheaper and cheaper models nowadays, and such emulated gyro can never get even close to smoothness and responsiveness of a real gyro. Besides that, we already know that it won't work with lots of interesting apps, especially VR ones.