frazew/VirtualSensor #33: Gyroscope not detected by apps

wujko, 2017-12-01:
Hello!
To begin with, I want to thank you for putting your time and effort into this project.

So, I bought myself a VR headset just to try it with my mobile, and I was pretty sure I have gyroscope. But it turned out, I don't. I've read then about Virtual Sensor and tried it out. So, it works, and in testing app like Sensor Box it is detected and works fine. However, no VR app, like e.g. Google Cardboard app, detects the emulated gyroscope. The only app that detects it is Sites in VR (which worked well without gyro), but it's amazingly shaky and totally unpredictable, even with Noise Filter installed in Xposed. Any ideas what can I do? Also, I've got no experience in writing apps, but in fact I'd like to learn something :)

Regards.

imtlmzamora, 2017-12-04:
I have the theory that it is because of the hook taking the wrong VR
library in some OS versions like lollipop as the cardboard app works just
with the recent VR SDK. I have tried to look more into it but I have in
this moment a lot of work.

wujko, 2017-12-05:
I forgot to mention the specs of my phone. It is LG G3s (or mini), with custom ROM LineageOS 14.1 (Nougat). If you'd like any more information or me to test something, just message me. In the end, I decided to buy a new phone, but still, I think this project is good for people with older phones that would like to try VR. So, if you need some specs, I can provide them :)

imtlmzamora, 2017-12-05:
for example, if I install the app cardboard 1.5 it works but not in the
version 1.8 so should be the google VR toolkit

Frazew, 2017-12-09:
Most likely the library has changed between versions indeed. We'd have to handle this.
I still don't have the time to work on this project though, so thanks for you help !

mirallas12, 2018-05-20:
hi. im also grateful for the efforts you've put into this hard project. i just wanted you to know that I have tried the virtual sensor on a rooted Samsung galaxy j5 prime sm-g570f/ds nougat 7.0 and it seems that the values for accelerometer is working but the gyroscope values aren't working. by the way i used the beta version so i know it could be buggy. i really wish i could help but im just a beginner in programming. thanks

louie123456789, 2019-01-02:
Im having the same problem with my lg stylo 2 plus on android 7.1.2 when it comes to vr because google cardboard doesnt detect any motion at all even tho it works in other apps. The custom rom im using is resurrection remix.