frazew/VirtualSensor #35: VirtuaSensor Not working on g530h

Voltonik, 2018-04-20:
Hello,
I have a problem with the Virtual Sensor module.
here is what I know:
the module is installed and enabled. But the Google cardboard app doesn't move. Some apps say I don't have gyroscope. And some say I have. YouTube 360 videos work. But when I switch to vr mode it doesn't move. Also google cardboard only moved on version 1.5. But only the demo moves. All apps that use gyroscope don't move.
And I have an idea for another module you can make.of course you know the var's vr player. It uses the compass as a gyroscope
and it works pretty well. So you can make a module that makes apps that use gyroscope work with compass.
Thanks..

Voltonik, 2018-04-22:
Well. The only problem is that all apps use the latest version of Google cardboard sdk. And the latest version of Google cardboard doesn't work....so either you can fix that. Or u can make another module that uses the compass as the gyroscope in Google cardboard apps..
Plz reply

tkrym02, 2019-05-22:
yes we need this please developer